bigtreetech/BIGTREETECH-Relay-V1.2 #1: [BUG] Issue in document translation

kamahat, 2019-12-10:
Hello,

On the document "Relay V1.2 Operating Instruction.pdf", on chapter IV, the schematic is in chinese and not in english. This is not helpfull to understand what port/connector is used for.

combolek, 2019-12-29:
I agree this needs to be fixed in the manual but in the meantime see this diagram:
[image]

kamahat, 2020-04-12:
> I agree this needs to be fixed in the manual but in the meantime see this diagram:
> [image]

Is this Wiring diagram all good ? 
the top left connector is labeled "Power failure detection diagnal line" as the last one but on the board it's a 5V power supply.

combolek, 2020-04-12:
Yeah, the translation is poor. The top "power failure detection" really means "short circuit detection". I have not used it so I am not sure how it works but see issue #5 for example connections that other people did. There is other useful info in that thread too.

But the bottom "power failure detection" has literal meaning, i.e. detection when power goes out.

carl1961, 2020-06-04:
I believe The top "power failure detection" it 5V in from the Mother Board, this may be used to feed the signal out on pin 3 , that a common procedure on many power resume modules. example pictures
[image]

[image]

roetherb, 2020-08-02:
Just a note to BTT if they see this: Please avoid creating diagrams with black backgrounds. I spent entirely too much time trying to remove it before giving up and wasting black toner just to get a print for an assembly workstation.