bigtreetech/BIGTREETECH-Relay-V1.2 #2: Auto off after 15 second

BaxMax80, 2019-12-12:
Hello,
I connected relay with BTT SKR MINI E3 V1.2 by PT-DET.
After around 15 second after I on my printer relay automatically turn off printer.
Is there any solve of this issue?

combolek, 2019-12-29:
I am not sure if this is your problem but the software shutdown function has quite a long delay (not sure exactly but on the order of 10 seconds) so maybe what you are seeing is the symptom of the shutdown signal pin being shorted. Check your connection to the pin marked "S" on the middle connector. Shutdown is activated (apparently) by setting this pin low for several seconds. It's not exactly documented as such so I may be wrong...

gonzaloalbito, 2020-01-10:
I'm also having the same problem.
I'd connected PC12 to S and GRN to G. I'd tryed with the 5V to 5V and GRN to GRN connection.
Applied all marlin changes from the manual, and added, including the missing PS_ON_PIN missing declaration in the pins_BTT_SKR_MINI_E3.h file.
Poor documentation doesn't help at all...

bryan-pl, 2020-01-27:
I'm also faced this problem. The board shuts off printer after 15 seconds after switching ON or resetting.

gonzaloalbito, 2020-01-28:
Mine shutdowns the printer even without being connected to the board, so I suspect it may be related to a defective relay.
It seems there are few more people with this problem. I've just read a reddit post from a guy with the same problem.
It seems BTT doesn't pay attention to their tickets or emails anyways.

bryan-pl, 2020-02-01:
Probably I've found the root cause of problem. Relay 1.2 needs HIGH level on input from mainboard to be active and LOW level for shut down. The input is not pulled to HIGH from Vcc by default, so with no signal on input module will shut down after 15s. Unfortunately SKR mini E3 1.2 is not setting the proper level on PT-DET after power on (even set properly in marlin configuration). The appropriate level on PT-DET is set after power off / on sequence set from display / controller.  I need to check the possible bug in marlin for Mini E3 to have it exact set.

jffabreu, 2020-02-13:
I have a Ender 3 with SKR v1.2, to make it work I just set the **PSU_ACTIVE_HIGH as FALSE**.

[image]

The relay signal pin must receive 5v to keep the printer ON, when it receives 0v it will wait for 10s ~ 15 seconds and shut the printer off.

The PSU_ACTIVE_HIGH parameter defines what is the default signal the board will send to the pin: FALSE = 5v and TRUE = 0v

The M80 sends the default signal to the pin and the M81 sends the opposite signal.

gonzaloalbito, 2020-02-15:
This is my configuration from the beggining but it's not working for me at all...
[image]

gonzaloalbito, 2020-02-15:
I'd been checking the PT-DET PC-12 voltage with both PSU_ACTIVE_HIGH=true and PSU_ACTIVE_HIGH=false. First thing I noticed is there is no 5v signal as someone suggested before.
When PSU_ACTIVE_HIGH=false then there are 0v on boot. When I power off manually from the menu then it gets about 3.3v and it returns to 0v when I power on manually again.
When PSU_ACTIVE_HIGH=true then there are 0v on boot. When I power off manually from the menu it still keeps 0v and it goes to 3.3v when I power on manually again.
It seems it always boots at 0v no matter which value you select here.

bryan-pl, 2020-02-15:
Even you have PSU_ACTIVE_HIGH=true and rest of Marlin configured OK, there is a bug in Marlin / SKR mini E3 behaviour after startup keeping the PS_ON_PIN low after startup. 
The workaround is below.
Go to MarlinCore.cpp file , find the function void setup(); //line 823
add within this function a  below line
PSU_ON();

I've added above line before end of setup() function - (line 1118);
Compile marlin and upload new firmware.bin file into SKR Mini E3. Then you have the expected behaviour of motherboard and relay.

gonzaloalbito, 2020-02-16:
> Even you have PSU_ACTIVE_HIGH=true and rest of Marlin configured OK, there is a bug in Marlin / SKR mini E3 behaviour after startup keeping the PS_ON_PIN low after startup.
> The workaround is below. [...]

It worked perfectly, thank you!!!

robus1, 2020-07-20:
I also faced with this problem. I used bryan-pl solution but nothing happens. Mayne I did something wrong Could you show me fragment of code which I should change? Tnx

flatcapper, 2020-07-22:
Hey, I'm not trying to change the subject, but what about connecting the relay to the tft35 e3 v3.0 monitor? It has an option to turn off printer after done and a power port for this option I think. However since so much of the btt upgrade parts have no documentation or very poor documentation, how is end user to know. I was watching a YouTube video somewhere and it mentioned that the relay could be connected to either mainboard or monitor. Any ideas? I'd prefer not to have to mess with firmware since I'm very new and not sure where to start or what to do. thanks,

combolek, 2020-07-22:
There is already a separate ticket for the TFT35 connection (issue #6).

flatcapper, 2020-07-22:
combolek, yep there sure is, however no one has answered the issue for the other user either. While waiting for you all to get around to possible suggestions, I went ahead and hooked the relay up and connected the jumper from middle power of relay that is for auto off, to the 2.pin.power option on.my tft35. Guess what happened? How about the printer powers up and then shuts down about 15 seconds later.  However the great news is my instant reset momentary button worked like a charm. Now if someone that knows how to make the relay work with the tft35 board me and question (=6) would greatly appriciate it. Also after reading all these comments it doesn't sound like I even want to mess with the firmware as its hit or.miss if it works. I'm way to new to start messing with the pins folder in addition to the config-h folder, then flashing the firmware.. So recomend away. Just before you point someone to another member asking a question, how about making sure someone had answered it first. Otherwise it makes no sense.

Davideq74, 2020-07-22:
Hello, I m newbie :)
I have an ender 3 pro with SKR Mini E3 v2.0 TFT35 v3.0 and Autopoweroff module.
I m not able to modify and compile firmware....is there anyone that can send me a working firmware.bin ? Please!
Thank you

flatcapper, 2020-07-23:
Hello David,
I am in the same boat as you are in. I have a ender 3 pro that I installed the btt skr mini e3 v2.0 mainboard and tft35 e3 v3.0., and want to connect the btt relay v1.2 automatic shutoff module. I didn’t want to have to reconfigure firmware which I here may not work either, and would like to just connect to the tft35 monitor. I was told to refer to your question #6 for answer. However after clicking on the link to this question, I found that no one had answered or supplied any potential answers to the issue. I don’t have any answers for you either. I installed the relay and can confirm that it does indeed shut off the printer 10-20 seconds after powering up. Sadly the plug that goes from relay to monitor isn’t receiving any signal to keep relay turned on. I did install a reset switch using a momentary pushbutton switch which does work to turn the printer back on. However it only turns off once again after 10-20 seconds. I’ve read several other posts about this problem and they were installing the relay connected to the mainboard with the same issue. The printer just shuts off after a interval of time each and every time. They said you have to go into the firmware and make changes to the psu control settings, and then re-flash the firmware onto the mainboard. Does this then change all the other settings that are working in the printer, or does this just change the section of firmware that you changed? I don’t know and not sure If I want to play around. Sorry I don’t have an answer for you, but it also doesn’t sound like anyone else does either. Good luck and please let me know if someone tells you how to get the relay to work.
Sincerely,
Perry P

alkis01, 2020-07-26:
Hi, I am having the same issue.10sec after powering the printer the BTT relay shuts it off.
The solution bellow did not work. Anyone has suggestions

> Even you have PSU_ACTIVE_HIGH=true and rest of Marlin configured OK, there is a bug in Marlin / SKR mini E3 behaviour after startup keeping the PS_ON_PIN low after startup.
> The workaround is below. [...]

flatcapper, 2020-07-26:
That's the other thing I found. The psu settings in the latest Marlin software were not at line 823, but were at line 1118. I guess that doesn't matter as it all depends on which version of marlin your working with. And it sounds like it still didn't work for you.

alkis01, 2020-07-26:
I feel a bit dumb, but found my problem. The cable that came with the BTT relay had it's wires reversed, so i swap them and the relay works fine

robus1, 2020-07-26:
Alkis you're great👍😀Cables in the connector should be crossed

Davideq74, 2020-07-26:
OMG! Are they mad?!?! I will try at the end of august after holidays... thank you for yours testa and infos!

flatcapper, 2020-07-29:
Hello robus1, did you also have to make changes in the firmware also. I assume that you plugged into the skr mainboard? I’m trying to plug into the tft35 as the relay v1.2 is supposed to also work that way. BTT wrote me back on AliExpress and said to plug into the ps-on port, however it’s a 3 pin port so which two ports do I plug the two pin plug into. They also said to change the firmware for the tft35 but they didn’t say what to change. I’m thinking not much help and if they would just simply make some basic documentation on how to install their products instead of making the customer do all the leg work. Someday somebody is going to come out with a 3d printer and upgrade parts that are easily plug n play just like on modern computers and make a killing in the 3d printer market. They could charge more and I wouldn’t care if I could decide one day I wanted to add a automatic shutoff feature and all I had to do was wire it in, plug the plugs in where they go and then activate the option from the touch screen. Now wouldn’t that be nice. Same with automatic bed leveling.. Well I’m glad you got your relay working. Mine’s still in a box collecting dust..
Happy trails,
Perry

robus1, 2020-07-30:
No didn't change firmware. I only change wires in connector. My o pinion about BBt accessories is the same. They are not plug'n'play like BBt said

Davideq74, 2020-08-03:
Hello! Has anyone firmware.bin working with poweroff module enabled? I tested voltage under 2 pin connector on mainboard bit have no 5 volts....

gonzaloalbito, 2020-08-03:
I did the MarlinCore.cpp and Configuration.h changes and works fine.

This is my Configuration.h:
[image]

And this is my MarlinCore.cpp
[image]

Please confirm the relay doesn't turn off your machine when not plugged in to the mainboard because one friend of mine received a defective one which was always turning off no matter the pin was up or down.

flatcapper, 2020-08-03:
Hello everyone, I hooked my relay up through the tft35 and it's working great now. BTT sent me the info I needed to get it to work. I plugged the 2 pin lead from the relay into the ps-on socket to the tft35. Then in the firmware  config.ini file made change to the ps-on to 1. Now at least when I power on the printer it stays on.I do here the relay clicking as it checks the power on the relay port. I will include photos from them. Not sure if it will turn off after print is done as I can't get my bltouch to work hence no prints. I've tried everything and everytime I power the printer after flashing the firmware it gets warnings that leveling is off, and low drift is off. However I've made sure in config.h that drift is uncommented (// removed), as well as auto-homing activated. I've tried with the plugs from bltouch plugged into the z-port, and also with the 3 pin plug into z-port and 2 pin into a stop. Tried with z_pin_port commented and un-commented with same results. I'll keep at it today.[Image]

[Image][Image]

[image]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F9137516%2F89199436-d69f2080-d5ae-11ea-9f6d-1cab85f253bc.png&data=02%7C01%7C%7C7f27d80ebf444687981108d837c242fc%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637320654766637392&sdata=FVsfkQtHF2%2FVfOULQGvdWoxnYRE8%2BJNv8X%2Bq4h%2BTysg%3D&reserved=0>

[image]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F9137516%2F89199485-e585d300-d5ae-11ea-8fab-d9271011b819.png&data=02%7C01%7C%7C7f27d80ebf444687981108d837c242fc%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637320654766647386&sdata=9UaPTSiUbiG8Rh%2F9N3nozT2r5S%2BPVAbinW9pct9mmfs%3D&reserved=0>

Davideq74, 2020-08-03:
2 pin on the tft35? Not on skr mini e3?
Instructions i found said to put 2 pin connector on skr mini e3... o_O

flatcapper, 2020-08-03:
Davideq74, yep the instructions I got were from btt specifically for the tft35. I didn't want to mess with the firmware for the mainboard, and besides, the tft35 is supposed to be able to handle the relay, filament sensor, wifi, and several other things I'm forgetting. If you look at the photos I've posted they were for the tft35. I can say as far as working. so far it does. Before I had the relay connected to the tft35 but plugged into the power socket which was a 2 pin. Made sense to me, but btt said to plug into the tft35 ps_on socket using the gnd, and pc12 pins. I just dont know if the printer shuts down after print. I'll get there if I can get my bltouch to work. Trying to find some pre-compiled firmware for the skr mini e3 v2.0 mainboard with bltouch.
Hope this helps.
Perry

kalawang2000, 2020-08-04:
thank you alkis01 , i was ready to give up till i found your comment about the cable and look at both boards. they could have put that on the instruction manual. 

got mine working on my skr mini e3 v2 

plugged the relay to the ps-on pin of skr mini(reversed the signal and ground on one end of the cable) then
uncommented the following in the configuration.h

#define PSU_CONTROL
#define PSU_NAME "Power Supply"
#if ENABLED(PSU_CONTROL)
#define PSU_ACTIVE_HIGH true  

be sure that that the pin number is correct
#define PS_ON_PIN                          PC13 (on skr mini e3 v2 its pc13, it might be different on your board)
[image]

hope this helps

BenoitGrou69, 2020-08-14:
> merci alkis01, j'étais prêt à abandonner jusqu'à ce que je trouve votre commentaire sur le câble et regarde les deux cartes, bas et voici les broches sont inversées. ils auraient pu mettre cela sur le manuel d'instructions.
> 
> le mien fonctionne sur mon skr mini e3 v2
> 
> branché le relais sur la broche ps-on de skr mini (inversé le signal et la masse à une extrémité du câble) puis
> décommenté ce qui suit dans la configuration.
> 
> #define PSU_CONTROL
> #define PSU_NAME "Alimentation"
> #if ENABLED (PSU_CONTROL)
> #define PSU_ACTIVE_HIGH true
> 
> assurez-vous que le numéro de broche est correct
> #define PS_ON_PIN PC13 (sur skr mini e3 v2 son pc13, il peut être différent sur votre carte)
> [image]
> 
> J'espère que cela t'aides

Hello ! I tested on my SKR E3 V1.2 and it doesn't work

Davideq74, 2020-08-17:
Hello, has anyone a compiled file firmware.bin  (modified for relay module 1.2) for SKR mini E3 v2.0 for Ender 3 pro without BLtouch?
I m not able to modify Marlin and recompile....
Thank u

Davideq74, 2020-08-17:
> Davideq74, yep the instructions I got were from btt specifically for the tft35. I didn't want to mess with the firmware for the mainboard, and besides, the tft35 is supposed to be able to handle the [...]

Hi, I can't see any photos...

kalawang2000, 2020-08-17:
> > merci alkis01, j'étais prêt à abandonner jusqu'à ce que je trouve votre commentaire sur le câble et regarde les deux cartes, bas et voici les broches sont inversées. ils auraient pu mettre cela sur le manuel d'instructions.
> > le mien fonctionne sur mon skr mini e3 v2
> > branché le relais sur la broche ps-on de skr mini (inversé le signal et la masse à une extrémité du câble) puis
> > décommenté ce qui suit dans la configuration.
> > #define PSU_CONTROL
> > #define PSU_NAME "Alimentation"
> > #if ENABLED (PSU_CONTROL)
> > #define PSU_ACTIVE_HIGH true
> > assurez-vous que le numéro de broche est correct
> > #define PS_ON_PIN PC13 (sur skr mini e3 v2 son pc13, il peut être différent sur votre carte)
> > [image]
> > J'espère que cela t'aides
> 
> Hello ! I tested on my SKR E3 Mini V1.2 and it doesn't work

did you fix the wiring? the pins on the skr relay is different from skr mini e3 v2.0

Davideq74, 2020-08-18:
> did you fix the wiring? the pins on the skr relay is different from skr mini e3 v2.0

I tested with multimeter 2 pins on SKR mini e3 v2 but no voltage detected...

flatcapper, 2020-08-18:
Hello everyone, I am removing the btt relay v1.2 permanently as a failed piece of hardware. It's just sad when equipment all from the same manufacturer does not work together. And if the consumer is able to get it to work they had to reprogram their printer, as well as re-wire the jumper plug between the relay and the mainboard.
Well I did all this and although my.printer stays on, however when I try to.heat either the hotbed or hotend my printer just instantly reboots. when it restarts if I again try to pre-heat either the hotbed or hotend it will once again reboot my printer. There is absolutely no excuse for a company to produce components like this that take so.much work.
si

[Capture d'écran 05/08/2020 à 3 14 56 AM]<https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F16840005%2F89334949-e95a4800-d6c9-11ea-84fe-6df0d085594a.png&data=02%7C01%7C%7C1502d6e5fc4849fe57db08d843746863%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637333514523311761&sdata=o1KfgQTnBdP2UcwFH1PP52FH3rOOmv7ysvUgSZPBYvA%3D&reserved=0>

Davideq74, 2020-09-09:
Hello, finally I solved that problem! :)
 First of all must verify cable connection, check if red and black wires coming from SKR MINI E3 are on the right side of plug.
Then modify /check two config files:

Marlin\Configuration.h

#define PSU_CONTROL
#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)

#define PSU_ACTIVE_HIGH true      // Set 'false' for ATX, 'true' for X-Box

or (depends on firmware version)

#define PSU_ACTIVE_STATE HIGH       // Set 'LOW' for ATX, 'HIGH' for X-Box

Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_V2_0.h

#define PS_ON_PIN                          PC13  // Power Supply Control

Compile and intall. I solved the problem and Relay module works fine!

I suggest to work on original Marlin Firmware and to customize it for yopur printer (not the source from BTT repository cause it gives many compiling errors...)

flatcapper, 2020-09-10:
Hello Davideq74,
I suppose I could re- assemble the relay v1.2 for the 4 time. I’ve not had any luck getting it to work, even when btt tech tells me where to connect it to my tft35. So I did as instructed, as well as making the changes needed in the tft35 firmware. Flashed the tft35, connected the 2 pin wire to where l was told to connect it. Powered on the printer and at least the thing stayed on. So decided to put the relay to the test. So selected something to print. Upon pushing the print button and my print bed turned on to begin heating my printer just shuts down. I mean its like someone pushed the reset button shutting down the printer, then it comes back on and reboots. I can not print with this relay connected to my printer. I also tried to connect to the mainboard (skr mini e3 v2.0), connected the relay 2 pin wire to the ps-on port I think. I would have to look. Also made the needed changes to the firmware and flashed the printer. Sadly, I experienced the same thing as when the relay was connected to my tft35. And I followed your instructions since I found the link where they told how to set up the relay. I don’t know what I’m doing different then those that are getting this relay to work. If you know anyone that wants one they are more then welcome to mine. Just pay me what I’m into it and they can have it. I do thank you for answering and am glad that the relay is working for you. I think that all these companies making upgrades for these 3d printers need to get their acts together and produce products that you don’t have to be a programmer in order to use their products.
Sincerely,
Perry P.

egusa999, 2020-09-11:
In my SKR mini e3 V1.2,
PT-DET PC-12 is not working.(I don't know why.)
so,I connect neopixel pin and change pin assign. then,it works correctly.
[image]
[image]
[image]
[image]
[image]

gleny91, 2020-10-23:
> Even you have PSU_ACTIVE_HIGH=true and rest of Marlin configured OK, there is a bug in Marlin / SKR mini E3 behaviour after startup keeping the PS_ON_PIN low after startup.
> The workaround is below. [...]

hi, can you explain it to me ? thanks

gleny91, 2020-10-23:
i can't find void setup. i found void setup kill pin

Thanks

gleny91, 2020-10-23:
I MANAGED.. THANKS ALOT

BCTANBC, 2021-01-07:
PLEASE MAKE SURE THE PSU PIN IS NOT REVERSED. IF IT IS REVERSED IT WILL MAKE IT SHUTDOWN EVEN AFTER YOU HAVE PRESSED THE BUTTON. MY CABLE WAS CAME REVERSED. STUPID MISTAKE.

th0547, 2021-01-27:
So I have got a Ender 3 with a SKR E3 mini 2.0 but I can't get the firmware to compile, anybody that can help me out?

dvan1954, 2021-12-14:
I have a BTT Octopus V1.2 and BTT Relay V1.2.   I am running KLIPPER so there is no configuration.h file to modify.  I have the same problem as others in that the relay shuts off after about 15 seconds.  To make matters even more confusing I am also using a BTT Mini UPS.  I removed the UPS but still have the same problem.   The Octopus has a dedicated connector for the relay and I have triple checked the wiring.  It is correct as far as I can tell.  I even tried reversing it but it didn’t help.   I tried changing the pin state but KLIPPER a wouldn’t start if I modified the pin state.
What do I need to do to allow this to function correctly on an Octopus V1.2 using KLIPPER?

pranzatelli, 2022-01-11:
> I feel a bit dumb, but found my problem. The cable that came with the BTT relay had it's wires reversed, so i swap them and the relay works fine

Just to second this for anyone that needs this page later: not only is the wire that comes with the kit swapped, [the image of proper board setup uses a swapped wire](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/blob/master/BIGTREETECH%20Relay%20V1.2/BIGTREETECH%20Relay%20V1.2/Hardware/%E5%BC%80%E5%85%B3%E5%8A%A0%E6%89%93%E5%AE%8C%E5%85%B3%E6%9C%BA.jpg). In a truly terrible twist, matching this image means you should take connect the wires in reverse.

GregNuspel, 2022-12-10:
I have found the Manta M8P loads too slow for the pin to go high in time to stop the shutdown from occurring. I have hooked up my relay so that it wont shutdown the system for a test. If I use the manual reset on the relay board it will reactivate. And my END_PRINT G-Code works as it should shutting the relay off after the extruder has cooled down. Is there any way to lengthen the delay of the auto shutdown of have the pin go high at startup in the firmware?

The Klipper pin config:

```
[output_pin PS_ON_OFF]
pin: PC3
value: 1

```
My shutdown G-Code is:
```
[gcode_macro END_PRINT]
gcode:
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Move nozzle away from print while retracting
    G91
    G1 X-2 Y-2 E-3 F300
    # Raise nozzle by 10mm
    G1 Z10 F3000
    G90
    G1 X10 Y330 F1000
    # Disable steppers
    M84
    G90 ; absolute positioning
    TEMPERATURE_WAIT sensor=extruder maximum=60
    SET_PIN PIN=PS_ON_OFF VALUE=0; shut down the printer

```

jariddler, 2023-01-13:
> I have found the Manta M8P loads too slow for the pin to go high in time to stop the shutdown from occurring. I have hooked up my relay so that it wont shutdown the system for a test. If I use the [...]

yes i have figured out a solution to this issue. well a work around to the main problem. if you have a button on the reset pins then hold it until klipper loads everything works as it should

Strimaud, 2023-03-14:
Hello, just installed btt relay v1.2 & just doesn't work. 
I get the shutdown after about 15sec. 
Runing skr3 ez with Marlin 2.1
Checked the wires: conduction ok, modified configuration.h:

#define PSU_CONTROL
#define PSU_NAME "Power Supply"
#if ENABLED(PSU_CONTROL)
#define PSU_ACTIVE_HIGH true
Still doesn't work. 
Check tension coming from pin pe4 with multimeter: surprise, got a 1.3v. Instead off 5v. Can't understand why?
Think I need some help please.

zardi125, 2023-03-20:
it keep tit tak tit tak

Strimaud, 2023-03-23:
Finally found the clue. For some reasons the dedicated ps on pin didn't fork so had to declare it in pins file, using the zstop pin (Gnd & Signal) which was unused (have a bltouch) and also works under 5V but could used any other pin matching 5v. 
Hope this may help. 
Cheers