bigtreetech/BIGTREETECH-Relay-V1.2 #4: Firmware source code

combolek, 2019-12-29:
Can you provide the source code for the v1.2 firmware ( BIGTREETECH-Relay-V1.2/BIGTREETECH Relay V1.2/BIGTREETECH Relay V1.2/Firmware/STC15W201S.hex )?

Whatever language it was originally written in is OK (8051 assembly language or C or whatever).

Thanks

Evg33, 2020-02-12:
```
CSEG AT 0000h
	LJMP L0001

L0008:
	SETB P3.0
	CLR P3.1
	RET

CSEG AT 000Bh
	DB 02h
	DB 00h
	DB 0Eh
	DB 0C0h
	DB 0E0h
	DB 0C0h
	DB 0F0h
	DB 0C0h
	DB 083h
	DB 0C0h
	DB 082h
	DB 0C0h
	DB 0D0h
	DB 075h
	DB 0D0h
	DB 00h
	DB 0C0h
	DB 00h
	DB 0C0h
	DB 01h
	DB 0C0h
	DB 02h
	DB 0C0h
	DB 03h
	DB 0C0h
	DB 04h
	DB 0C0h
	DB 05h
	DB 0C0h
	DB 06h
	DB 0C0h
	DB 07h
	DB 020h
	DB 0B3h
	DB 011h
	DB 07Fh
	DB 032h
	DB 07Eh
	DB 00h
	DB 012h
	DB 00h
	DB 07Eh
	DB 020h
	DB 0B3h
	DB 07h
	DB 0D2h
	DB 0CDh
	DB 012h
	DB 00h
	DB 03h
	DB 080h
	DB 0FEh
	DB 0D0h
	DB 07h
	DB 0D0h
	DB 06h
	DB 0D0h
	DB 05h
	DB 0D0h
	DB 04h
	DB 0D0h
	DB 03h
	DB 0D0h
	DB 02h
	DB 0D0h
	DB 01h
	DB 0D0h
	DB 00h
	DB 0D0h
	DB 0D0h
	DB 0D0h
	DB 082h
	DB 0D0h
	DB 083h
	DB 0D0h
	DB 0F0h
	DB 0D0h
	DB 0E0h
	DB 032h
L0003:
	MOV 0CAh, #0FFh
	LCALL L0004
	LCALL L0005
	MOV R7, #40h
	MOV R6, #1Fh
	LCALL L0006
L0007:
	JNB P3.2, L0007
	MOV R7, #4Ch
	MOV R6, #1Dh
	LCALL L0006
	JNB P3.2, L0007
	SETB 0CDh
	LCALL L0008
	SJMP L0007

L0006:
	MOV A, R7
	DEC R7
	JNZ L0009
	DEC R6
L0009:
	DEC A
	ORL A, R6
	JZ L0010
	NOP
	NOP
	NOP
	MOV R5, #0Bh
	MOV R4, #0BEh
L0011:
	DJNZ R4, L0011
	DJNZ R5, L0011
	SJMP L0006

L0010:
	RET

L0001:
	MOV R0, #7Fh
	CLR A
L0002:
	MOV @R0, A
	DJNZ R0, L0002
	MOV SP, #8h
	LJMP L0003

L0004:
	CLR 0CDh
	CLR P3.0
	CLR P3.1
	SETB P3.2
	LJMP L0012

L0005:
	MOV 8h, #1h
	SETB IT1
	SETB EA
	SETB EX1
	RET

L0012:
	CLR P3.0
	SETB P3.1
	RET

END

```

Evg33, 2020-02-12:
P3.0 and P3.1 firmware upload uart
P3.2 shutdown
P3.3 short circuit
P5.4 reset/restart
P5.5 relay

[image]

combolek, 2020-02-12:
It seems that some disassembly above is incomplete. Here is slightly better version.
[STC15W201S.txt](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/files/4195775/STC15W201S.txt)

combolek, 2020-02-13:
Still, I a real source code would be desired. It's not like there are some trade secrets there. So we should not have to reverse engineer it.

PapiOrso, 2020-02-18:
BTT sells defective components and is unable to respond to customers

sabellnikov, 2022-10-21:
Is there a way to increase turn off timer after started up?

sobieh, 2022-10-21:
[](https://github.com/sobieh/bttrelay)
You can use this one. I modified it to support TURN OFF and TURN ON without need to reset.

sabellnikov, 2022-10-21:
@sobieh thank you so much! I'll be very grateful for you if put more description how to flash it! Thanks!

sobieh, 2022-10-21:
> @sobieh thank you so much! I'll be very grateful for you if put more description how to flash it! Thanks!

As the readme says:
- Install python 3.x
- pip install stcgal
- Connect your board to any USB to Serial adapter
- Run upload.bat COM# where # is your COM port number
- Reset the board with reset button

It's really really easy... just connect it with breadboard wires like:
- RX -> TX
- TX -> RX
- GND -> GND
- 5V -> 5V

sabellnikov, 2022-10-21:
@@sobiech i'm doing everything righ (connection and etc) but powershell doesn't find the board. do I need any specifield drivers?

sobieh, 2022-10-21:
@sabellnikov seems to be related to your usb adapter or missing drivers

combolek, 2022-10-21:
Check the port number in Device Manager, maybe it's not COM6:

[image]

sabellnikov, 2022-10-22:
Yes. the pc doesn't find the device out...  is there any drivers? or it should work automatically?

combolek, 2022-10-22:
This depends on the particular chip in your USB->Serial adapter. Some have drivers included with Windows and some require additional drivers installed.

brutzler, 2023-03-25:
> https://github.com/sobieh/bttrelay You can use this one. I modified it to support TURN OFF and TURN ON without need to reset.

Can anybody implement a delay function for "power off"? This is not really my program language :-(
Want to use it in my Manta8P. If I switch it immediately off, then I make a hard shutdown of Klipper-OS on the integrated "RasPi" 
Would be a charme, if i could say to the module "power off in e.g. 30s" and then there is enough time to shutdown Klipper by software.

sobieh, 2023-03-25:
> Can anybody implement a delay function for "power off"? This is not really my program language :-( Want to use it in my Manta8P. If I switch it immediately off, then I make a hard shutdown of [...]

Is there any point in shutting down klipper by your Manta ? The purpose of this BTT switch mod is to be controlled by RPI and able to remotely turn ON/OFF the printer power and keep RPI always ON on a separate smaller psu. Klipper is "the one who knocks" :D

Currently the firmware it only waiting 3 sec. between checks if the Power pin is High or Low. It will not wait 3 sec after you change state 0->1 or 1->0. It only prevents you from repeated fast switching. You would have to redesign it to always wait 30 seconds if the state transition is 1 -> 0 (turn off). This is quite simple just add:

`
    mov     r7, #030h   ; delay(30000) 30 sec
    mov     r6, #075h
    lcall   delay
`

before line

`
setb    p5_5        ; turn off 230v
`

brutzler, 2023-03-26:
Thx for your answer. My english is not so good to understand all very well.
Fact is, that klipper is not powered by a secondary power supply. It is powered by the 24V (24V->5V) on the Manta-Board. 
This means, that klipper can not always be the master, because after loosing 24V supply, klipper is dead..
So my thoughts are:  After idle-state of the printer, I send a signal to the relay module to switch off after 30sec, and simultanly send a signal to klipper to shutdown. This sets the printer into an "OFF-MODE".
After waking up with the reset-switch, klipper is rebooting to normal state..
--> I will try your code-change.  Hope I am able to get the modified FW on the relay module.

brutzler, 2023-03-26:
This way?
---------
main_loop_on:
    jb      p5_5, main_loop_continue

    mov     r7, #030h   ; delay(30000) 30 sec
    mov     r6, #075h
    lcall   delay

    setb    p5_5        ; turn off 230v
    lcall   set_rx
------------------------

brutzler, 2023-08-13:
> Is there a way to increase turn off time (15 sec) after started up?

After some tests with the relay module, I see, that my only need is to increase the 15s (or is it 16s?)  delay after startup to a higher value.
I found the code for the 8s delay on switch off in the HEX-File  "7F 40 7E 1F" but where is the code for the delay after Startup?
I expected a "7F 80 7E 3E" for 16sec.  
The only second delay routine is "7F 4C 7E 1D" and this are 7.5sec....
Stucking here...

talv2010, 2023-10-14:
I did same firmware with delays. You can use it.
https://github.com/talv2010/biqu-relay-v1.2-firmware-modified

tiagomaricate, 2024-02-02:
Hi people!
Does somebody have a little tutorial about how to write the new firmware in the BTT Relay MCU?
I bought two piece of BTT Relay v1.2 and both simply doesn't work. It's like there is no firmware on it.
It simply turn ON and doesn't turno OFF. I can short-circuit the reset pins and the relays continues ON.
Any signal I put in pins does nothing. And both LEDs from relay boards continues ON.
After 20 seconds it doesn't turn OFF. As I said, simply do nothing.
So I'd like to try to flash a new firmware like the one from @sobieh.
I'm using a Mac, have a RS232 to TTL  with a USB cable connected in the "dev/cu.usbserial-210" port.
I was trying to use the CoolTerm app to write but not sure how to proceed.
I'm completely newbie on this matter to use serial connection to write in a MCU.
Appreciate any help!

combolek, 2024-02-02:
AFAIK, you need special programming software. You can get the Windows executable from https://www.stcmicro.com/rar/stc-isp6.91.rar
I don't know if there is a Mac version.

tiagomaricate, 2024-02-02:
> AFAIK, you need special programming software. You can get the Windows executable from https://www.stcmicro.com/rar/stc-isp6.91.rar I don't know if there is a Mac version.

Shell I manage to make a progress here.

Instead of use the RS232 to TTL now I'm using the BTT Writer board.
Also I undesrtand (at least some parts) of CoolTerm and did a test flash the firmware from @sobieh .
The good news is that now the really is partially working.
It reset and disarm after few seconds.
The bad part is not powering my printer.
Not sure if is possible to diferenciatte 110v and 220v in firmware. If so, I have to discover how to modify it to 110v.

tiagomaricate, 2024-02-03:
Oh, my mistake. About pouring printer it was a bad connection.
Now it's working.
But one of the relays is not flashing or something similar.
Doing the same thing with both, one of them continues not working.

Escrich, 2024-02-03:
> Oh, my mistake. About pouring printer it was a bad connection.
> Now it's working.
> But one of the relays is not flashing or something similar.
> Doing the same thing with both, one of them continues not working.

I'll try a bit before to try to reload firmware, what happens exactly?

combolek, 2024-02-03:
> Not sure if is possible to diferenciatte 110v and 220v in firmware. If so, I have to discover how to modify it to 110v.

I don't think the relay cares. All the control logic is running on the low voltage. The load can be anything up to the maximum spec.

Escrich, 2024-02-03:
The black supply module is intended to work between 100 and 250 vac, mains input voltage is not your problem

tiagomaricate, 2024-02-03:
Well, so far the results are:

1) BTT Relay (unit A): flashed successful. The relay is back to work normally. But once I plugued the short circuit detect connector, the IC exploded. I have a replacement part but as @Escrich wrote in another topic, this can be removed and replaced by a "bypass" jumper and the M80 printer command can back to work.

[image]

2) BTT Relay (unit B): I followed the same process to flash it but so far there is no difference in behavior. After I turn ON the power, the really continues ON (with both LEDs ON) and simply doesn't turn off. Any connection or 5V sinal makes no difference.The relay continues static and ON.

So I will need some help to use another tool to try flash it again.

Escrich, 2024-02-03:
how do you make the short circuit detection?

tiagomaricate, 2024-02-03:
> how do you make the short circuit detection?

There is a contact to use a jumper close to the conector. If you use the jumper, the sc detection is active.

Escrich, 2024-02-03:
There's no way to make the chip explode just closing the jumper, maybe you're doing something wrong, I have a couple of relays, and I don't use them, for some reason, but them work, are so simple, as simple as it's almost impossible to fail

galagithub, 2024-12-05:
New open source firmware written in C:
https://github.com/galagithub/btt_relay_os

Not affiliated with BTT.

galagithub, 2024-12-05:
> There's no way to make the chip explode just closing the jumper, maybe you're doing something wrong, I have a couple of relays, and I don't use them, for some reason, but them work, are so simple, as simple as it's almost impossible to fail

The 5V_IN goes directly to the LM358 OPAMP with no protection. In theory, this will handle VCC+2V on it's inputs, but it is not a rail-to-rail OPAMP so the circuit has a HW bug in it. Do not use this input, unless you can be sure that the voltage on it is less than 7V, even then it is stretching the limits.

Escrich, 2025-04-11:
Finally I did my own using an Attiny85 MCU, much easier to be handled, instead to use an operational amplifier for short circuit detection, I was used a common analog input on Attiny, ready to detect short circuits even in a 3'3V power systems, short circuit test is selectable as in the original from BTT, also there is an output for power out, an enhanced reset push button input, and of course the control input from Klipper or Marlin board, in this case, initial start up time is 90 seconds, and time to switch off once started is 10 seconds, when selected, the short circuit detection will trip the system in two seconds under any circumstances, even if we are inside the start up time.

[image]

Escrich, 2025-04-11:
> Finally I did my own using an Attiny85 MCU, much easier to be handled, instead to use an operational amplifier for short circuit detection, I was used a common analog input on Attiny, ready to detect short circuits even in a 3'3V power systems, short circuit test is selectable as in the original from BTT, also there is an output for power out, an enhanced reset push button input, and of course the control input from Klipper or Marlin board, in this case, initial start up time is 90 seconds, and time to switch off once started is 10 seconds, when selected, the short circuit detection will trip the system in two seconds under any circumstances, even if we are inside the start up time. A blue led is showing us the system status, changing the flashing speed due to the current status
> 
> [image]

Of course you can test or edit, without having the 220V present, just using the power loss output as power input, that's another system advantage

Escrich, 2025-04-11:
The PCB:

[image]