bigtreetech/BIGTREETECH-Relay-V1.2 #5: SKR 1.3 connection

abdbeg, 2020-01-03:
Hi

I was tried several ports and combinations which took about 3 hours but the relay doesn't work with m81 command, can somebody share their skr 1.3 used pins and firmware changes for this relay? 

Thanks in advance!

combolek, 2020-01-05:
I don't have skr 1.3 but I am trying it with skr pro 1.1 and cannot get it to work either. But the relay module works fine when controlled by Raspberry Pi so I know it's not broken.
One thing I did notice was that for the software shutdown to work at all the shutdown signal must be set to low for at least 8 seconds. Not sure if this is expected.

abdbeg, 2020-01-05:
I was finally was able to make it work, firstly i used thermistor pins instead of endstops, the main thing is that S pin from the relay should be connected to the positive board pin ( 0.25 in my case, the pin should be able to send signal, this is very important because S pin in endstop can't do that and that's why it didn't work).

combolek, 2020-01-05:
Hmm, I was trying different pins and on SKR PRO they are quite generic 
and endstop pins can send signal if you configure them for output. I 
verified with multi-meter that the signal is being sent.

Do you also see a delay or does it shutdown immediately when you send 
the signal?

abdbeg, 2020-01-06:
Hm, I don't have multi-meter so I'm not sure about sending a signal, also no, it doesn't shut down immediately, it takes about 5 seconds to turn off.

CristianoGorlaBDF, 2020-01-11:
I did the same, I tried the pin 1.26 ad S but it didn't work, then I did a test with the multimeter.
The measure was strange, ~0.4v with S=0 and ~0.6v with S=1 with the relay connected. Then, i did a check without the relay connected and it was 0v with S0 and 5v with S=1.
Then i tested it with Arduino and all was as espected.
Looking at the schematic the only difference I can see is a resistor between the pin and the mcu.
I tested also the short circuit detection and it works.

juan6627, 2020-01-13:
> I was finally was able to make it work, firstly i used thermistor pins instead of endstops, the main thing is that S pin from the relay should be connected to the positive board pin ( 0.25 in my [...]

Thank you for sharing your experience, your comment helped me get it to work on my skr v1.3.

CristianoGorlaBDF, 2020-01-13:
> Thank you for sharing your experience, your comment helped me get it to work on my skr v1.3.

You are welcome, I forgot to mention that the relay v1.0 works with the endstop pins.

sbaydar, 2020-01-23:
> Thank you for sharing your experience, your comment helped me get it to work on my skr v1.3.

Could you share wiring diagram and marlin modification for SKR V1.3 ? 
 my email is soner_baydar@hotmail.com
Thank you

abdbeg, 2020-01-23:
I will publish it here in case if someone needs it.

I attached the wiring, the only difference is that I used thermistor pins instead of endstop.

[image]

The marlin configuration you can find in the PDF manual, you need to change pins for your (P0_25 in SKR 1.3 if you are using the second thermistor) and... that's it.

CristianoGorla, 2020-01-23:
The red connectin is for the shutdown signal, the light blue for the 5v short circuit detection (i don't know your configuration so i marked different spots where you can connect 5v circuit, choose just one). On the yellow spot on the relay put a jumper cap and a momentary push button (NO) on the green spot.
[image]
For the firmware follow the manual chapter VI (change just the pin number, if you followed my wiring put P0_25).

abdbeg, 2020-01-23:
@CristianoGorla wow thanks, I didn't know about the jumper, do you need to remove it after pushing the button?

sbaydar, 2020-01-23:
> I will publish it here in case if someone needs it.
>
> I attached the wiring, the only difference is that I used thermistor pins instead of endstop.
>
> [image]
>
> The marlin configuration you can find in [...]

I checkhed on pdf but this example for skr  pro. I need for SKR V1.3 and relay v1.2

[image]

CristianoGorla, 2020-01-23:
> @CristianoGorla wow thanks, I didn't know about the jumper, do you need to remove it after pushing the button?

No, ther cap enables the short circuit detection, the push button is needed to rearm the relay after the M81 gcode.

CristianoGorla, 2020-01-23:
> I checkhed on pdf but this example for skr pro. I need for SKR V1.3 and relay v1.2
>
> [image]
From my previous message:
> For the firmware follow the manual chapter VI (change just the pin number, if you followed my wiring put P0_25).

sbaydar, 2020-01-23:
> llow spot on the relay p

Thank you so much . How can i find of the pins name? 

[image]

CristianoGorla, 2020-01-23:
In the previous comment just below the pic you reposted:
[image]
or i my first comment:
[image]

sbaydar, 2020-01-23:
> In the previous comment just below the pic you reposted:
> [image]
> or i my first comment:
> [image]

Thank you for your answer . I want to ask about extruder cooling. Can we setup extruder cooling degree before automatic shutdown ?

CristianoGorla, 2020-01-23:
Sure 
M109 R50 ; Set nozzle temperature to 50°C and wait till reach it
M81 ; Shutdown

sbaydar, 2020-01-23:
> Sure, you should edit your end gcode:
> 
> M109 R50 ; Set nozzle temperature to 50°C and wait till reach it
> M81 ; Shutdown

Thanks a lot

Vik31, 2020-02-08:
My configuration:

-BTT SKR1.3 with double extruder, so the pin 0_25 of TH1 is for thermistor of second hot-end, can't be used to drive the relay V1.2
-BL-Touch so the servo pin 2_0 can't be used to too
-TFT35 screen: EXP1 and EXP2 plug are free 

To drive the relay V1.2 is necessary minimum 2.6Vdc
I found pin 0_28 is good to drive relay BTT V1.2

I used GND on the opposite corner of EXP1 plug

[image]

[image]

[image]

[image]

mralko99, 2020-04-06:
> I don't have skr 1.3 but I am trying it with skr pro 1.1 and cannot get it to work either. But the relay module works fine when controlled by Raspberry Pi so I know it's not broken.
> One thing I did [...]

How do you controlled it with the raspberry pi

combolek, 2020-04-06:
> How do you controlled it with the raspberry pi

You have to connect ground and two GPIO pins from RPi to the relay. I use pins 6 (GND), 8 (TXD), 10 (RXD). This is maybe somewhat confusing because TXD and RXD are normally used for UART but I don't need it so I just use them as GPIO. You can choose different pins.

6 goes to GND on the relay, 8 goes to relay shutdown signal (S pin on the middle connector) and 10 goes to RESET pin on the relay.

Then I use this python script to turn the printer on and off (call it with "on" or "off" argument):

```python
#!/usr/bin/python

import RPi.GPIO as GPIO
import sys
import time

PS_ON_PIN =  8
RESET_PIN = 10

GPIO.setmode(GPIO.BOARD)
GPIO.setwarnings(False)
GPIO.setup(PS_ON_PIN, GPIO.OUT, initial=GPIO.HIGH)
GPIO.setup(RESET_PIN, GPIO.OUT, initial=GPIO.LOW)

if sys.argv[1] == "on":
    GPIO.output(PS_ON_PIN, GPIO.HIGH)
    GPIO.output(RESET_PIN, GPIO.HIGH)
    time.sleep(0.5)
    GPIO.output(RESET_PIN, GPIO.LOW)
elif sys.argv[1] == "off":
    GPIO.output(RESET_PIN, GPIO.LOW)
    GPIO.output(PS_ON_PIN, GPIO.LOW)
    time.sleep(8.0)
    GPIO.output(PS_ON_PIN, GPIO.HIGH)
else:
    print "invalid operation"
    GPIO.cleanup()
    exit(1)

GPIO.cleanup()
```

bryan065, 2020-07-21:
If anyone else is curious, that last connector is a 5v short circuit test point. Always supplying ~4.75-5v even when relay is off. I've used that port to hook up to the power pin on the BTT TFT35 to enable always on tft even when relay is off. (Good for doing M80)

KL-XL, 2020-08-02:
> On the yellow spot on the relay put a jumper cap and a momentary push button (NO) on the green spot.

Hi - where can I get the jumper from please? I am pretty sure the relay came without it. Would the red jumpers that came with my SKR Mini E3 board for sensorless leveling work?
Thanks

combolek, 2020-08-02:
Yes, these should work.

KL-XL, 2020-08-04:
Hi - me again :) 
Apologies if my questions are too dumb - I have a high-school level of understanding of electricity :)

I’d like to add a momentary reset-button with LED to my relay. What button should I get and how do I power the LED up? An answer for dummies and/or a link to an actual product (AliExpress, Amazon, eBay, etc.) highly appreciated :)

Many thanks in advance.

**My configuration is as follows:**
* Ender 3
* BTT SKR Mini E3 2.0
* BTT TFT35 E3 3.0
* BLTouch 3.1
* BTT UPS 24V 1.0
* BTT Relay 1.2
* BTT SFS 1.0
* Mean Well LRS-350 24V 14.6A

Also awaiting delivery of
* Raspberry Pi 4B with cam
* 2 x Pololu 5V/3.2A Step-Down Voltage Regulator (D36V28F5), one for Raspberry Pi and one spare

All electrics and electronics are moving to a 3D-printed (PETG) external case.

combolek, 2020-08-04:
You can use something like this: https://www.aliexpress.com/item/886978271.html (the momentary 24V version). It has 5 contacts/wires. Connect the `+` and `-` wires to the `V+` and `V-` output terminals of the PSU (so the LED is on when the PSU is on). Connect the `C` wire to the `5V` pin on the relay reset header and the `NO` wire to the `RST` pin on the reset header. Leave the `NC` wire unconnected.

KL-XL, 2020-08-04:
Thank you. i understand those buttons have a maximum current of 3A and the PSU has up to 14.6A. Is that not going to be a problem? Will it not fry the diode?

combolek, 2020-08-04:
I do not see the 3A anywhere but this must be the rating of the switch itself. The actual current is going to much less because it's only connected to the relay. The switch and the LED are completely separate (not electrically connected), they are just in the same housing. And the LED is 24V rated (this is what the choice on aliexpress means, the switch part doesn't care about the voltage). LED will only use some miliamps. The max current rating of the PSU is how much it can deliver at most but the actual current is always dependent on the load (unless you have a short or overload the PSU).
To be honest however, if you don't understand about electricity you should not be mucking with this. You are dealing with high voltage stuff that is dangerous and can kill you, start a fire or worse.

KL-XL, 2020-08-04:
Thanks very much. Really appreciate your time and advice. You are right - this item doesn't specify the current. I looked at a bunch of similar buttons and they did specify 3A either in the Item Description or Specifications. I also asked one of the sellers and they literally said this: _"as long as the voltage corresponds, the current can be directly connected if it does not exceed 3A"_ (https://www.aliexpress.com/item/4000375272251.html). I must have misinterpreted it, it can obviously mean exactly what you already said about the load.
On the other topic, you probably can tell that the reason why I am asking dumb questions is I am very conscious of the risks and am taking them very seriously. I am going very slowly and getting multiple views before I touch anything in my printer. I do have the basic electrical precautions habits as I have been assembling my custom PC's for decades.

HTTYD9, 2020-08-27:
> I was finally was able to make it work, firstly i used thermistor pins instead of endstops, the main thing is that S pin from the relay should be connected to the positive board pin ( 0.25 in my [...]

can you please send diagram for skr pro v1.1

sergd1970, 2020-08-27:
Hi. I have a problem with BIGTREETECH V1.2, after turning on the power, it works for 20-25 seconds and then turns off.  What can be done?
Sir 1.4 turbo board. 
 The detector for a short circuit is connected to the terminal X, and the signal a power off is to the thermistor 2 of the hot end

wolfgangmauer, 2020-08-28:
Remove any M80 in your configuration/startgcode, it don't work!

den180276, 2020-09-16:
Hello everyone!
I need help.
Who set up the module in Klipper?

combolek, 2020-09-16:
I have it working with Klipper on SKR 1.4 Turbo. I connected it to PWRDET header. In printer.cfg I have this:
```
[output_pin ps_on]
pin: P1.0
value: 1

[gcode_macro M81]
gcode:
    set_pin pin=ps_on value=0
```
Then M81 works as expected and shuts down the PSU (there is about ~8sec delay though).

ghdtnsld, 2020-09-18:
> Hi. I have a problem with BIGTREETECH V1.2, after turning on the power, it works for 20-25 seconds and then turns off. What can be done?
> Sir 1.4 turbo board.
> The detector for a short circuit is connected to the endstop connector X, and the signal a power off is to the thermistor 2 of the hotend

I am in the same situation.  Have you solved it?

wolfgangmauer, 2020-09-18:
@ghdtnsld Don't you like the solutions described above?
Forget the short circuit first. "Power off" to a connection where a switch would also work,
in youre case "endstop connector X" change firmware settings, and everything works.

PS: Connecting a Power down output to a Thermistor Input sounds a little strange, isn't it?

ghdtnsld, 2020-09-18:
> @ghdtnsld Don't you like the solutions described above?
> Forget the short circuit first. "Power off" to a connection where a switch would also work,
> in youre case "endstop connector X" change [...]

That was the answer!  Thank you!

olivbd, 2020-11-27:
I found the schematics on a russian website... yes it's reverse-engineering!

[image]

Sources:
- https://3dtoday.ru/blogs/vasilius-v/reversing-modulya-rele-bigtreetech-relay-v12/
- https://github.com/Vasilius-001/BIGTREETECH-Relay-V1.2

ricalv1973, 2020-12-18:
Hello, anyone got this working on a GTR V1.
The relay is on, but doesn't respond to M80 / M81 commands.
I have tried connecting the 5v+gnd to the I2C pins and the S+g to the PH6 connection, but still doesn't respond.

olivbd, 2020-12-18:
> Hello, anyone got this working on a GTR V1.
> The relay is on, but doesn't respond to M80 / M81 commands.
> I have tried connecting the 5v+gnd to the I2C pins and the S+g to the PH6 connection, but still doesn't respond.

Check this: #7

sinole1987, 2020-12-24:
I am using that thermistor header for my pinda 2. Can I do below configuration to save headers.
[image]

softwarecrash, 2021-02-17:
> You have to connect ground and two GPIO pins from RPi to the relay. I use pins 6 (GND), 8 (TXD), 10 (RXD). This is maybe somewhat confusing because TXD and RXD are normally used for UART but I don't [...]

that doesnt work, the pin from the RPI have not enugh power to hold the on signal

combolek, 2021-02-17:
> that doesnt work, the pin from the RPI have not enugh power to hold the on signal

It works for me, I use it all the time. Maybe it depends on the Pi version? (I use the original 4B)

combolek, 2021-02-17:
> > that doesnt work, the pin from the RPI have not enugh power to hold the on signal
> 
> It works for me, I use it all the time. Maybe it depends on the Pi version? (I use the original 4B)

Actually, I take it back. It appears that something broke and it's not working anymore. I didn't notice because I forgot that I **also** have the relay connected to SKR 1.4 so I can use M81 gcode in klipper (see https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/5#issuecomment-693698837) and that's how I normally turn the printer off.

You maybe right about the power but I am pretty sure it used to work. Confused.

softwarecrash, 2021-02-17:
> 
> 
> > > that doesnt work, the pin from the RPI have not enugh power to hold the on signal
> > 
> > 
> > It works for me, I use it all the time. Maybe it depends on the Pi version? (I use the original 4B)
> 
> Actually, I take it back. It appears that something broke and it's not working anymore. I didn't notice because I forgot that I **also** have the relay connected to SKR 1.4 so I can use M81 gcode in klipper (see [#5 (comment)](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/5#issuecomment-693698837)) and that's how I normally turn the printer off.
> 
> You maybe right about the power but I am pretty sure it used to work. Confused.

ok it work, the issue was GPIO.cleanup() it release all gpio to floating state.
a way to script this in bash? so that the psu controll plugin from octoprint can controll it?

combolek, 2021-02-17:
Ah, so I broke it myself :-1:  Thanks, will have to try this.

I am not familiar with PSU control plugin but you can certainly call the python script in a bash script.

Alternatively, the Enclosure plugin lets you call scripts or even control GPIO directly. There are other GPIO plugins too.