bigtreetech/BIGTREETECH-Relay-V1.2 #7: M80 not working with 1.2, but wor

Evg33, 2020-02-12:
G-code M80 not working with V1.2, but working with V1.0.

bryan-pl, 2020-02-15:
Because v1.2 is only sensitive on power-off (LOW) signal :(

Evg33, 2020-02-17:
Need a STC15 firmware modification ?

LyWill, 2020-07-09:
Any one think it's possible to simply connect the RST pin to a pin on the SKR1.3 and make the pin go HIGH to turn on the power supply? The 5+ pin next to the RST pin registers as 4.68V, which is lower than 5V. Wonder if 3.3v from the SKR will be enough to trigger the rearm.

LyWill, 2020-07-11:
I've rigged up a 2n 3806 to trigger the reset and it seems to work. I have the Base connected to a 1K resistor where I plan to hook up to the Z endstop pin, the Collector to the +5v, and the Collector to the RST pin. I plan to modify the firmware so M80 invokes the RST pin with the transistor and M81 shuts down the PSU.

LyWill, 2020-07-16:
OK my comment from before works with a tweek. For those that are interested, here is what I did.

**BOM:**

- 2N 3806 Transistor (PNP)
- 2N 3804 Transistor (NPN)
- 10K Resistor
- 1K Resistor

**Wiring**
Wire the circuit as described in this photo. The load in this case the RST pin.
[image]

**Marlin Code Modification**
pins_BTT_SKR_V1_3.h
The pin number will be different depending upon the board you are using. Mine is SKR1.3
Add
#define PS_ON_PIN P1_24
#define PS_OFF_PIN P0_25

MarlinCore.h

Search for 'PSU_CONTROL'

Change (or comment out with double slash)
  #define PSU_PIN_ON()  do{ OUT_WRITE(PS_ON_PIN,  PSU_ACTIVE_HIGH); powersupply_on = true; }while(0)
  #define PSU_PIN_OFF() do{ OUT_WRITE(PS_ON_PIN,  !PSU_ACTIVE_HIGH); powersupply_on = false; }while(0)
TO
  #define PSU_PIN_ON()  do{ OUT_WRITE(PS_OFF_PIN,  PSU_ACTIVE_HIGH); OUT_WRITE(PS_ON_PIN,  PSU_ACTIVE_HIGH); safe_delay(1000); OUT_WRITE(PS_ON_PIN,  !PSU_ACTIVE_HIGH);   powersupply_on = true; }while(0)
  #define PSU_PIN_OFF() do{ OUT_WRITE(PS_OFF_PIN,  !PSU_ACTIVE_HIGH); powersupply_on = false; }while(0)

This should allow you to do a M80 to turn on the relay and then a M81 to turn off the relay.
If you noticed in the code the M80 does not keep the PS_ON_PIN high the entire time, it goes high and then waits a second and then goes low. This is to simulate a 'button' press.

Evg33, 2020-07-16:
Easier to buy the old version 1.0 from banggood.

bryan065, 2020-07-21:
> OK my comment from before works with a tweek. For those that are interested, here is what I did.
> 
> **BOM:**
> 
> * 2N 3806 Transistor (PNP)
> * 2N 3804 Transistor (NPN)
> * 10K Resistor
> * 1K Resistor
> 
> **Wiring**
> Wire the circuit as described in this photo. The load in this case the RST pin.
> [image]
> Honorable mention to the Youtube video that made this all possible:
> https://www.youtube.com/watch?v=UdAnUc7nXYs&t=31s
> 
> **Marlin Code Modification**
> pins_BTT_SKR_V1_3.h
> The pin number will be different depending upon the board you are using. Mine is SKR1.3
> Add
> #define PS_ON_PIN P1_24
> #define PS_OFF_PIN P0_25
> 
> MarlinCore.h
> 
> Search for 'PSU_CONTROL'
> 
> Change (or comment out with double slash)
> #define PSU_PIN_ON() do{ OUT_WRITE(PS_ON_PIN, PSU_ACTIVE_HIGH); powersupply_on = true; }while(0)
> #define PSU_PIN_OFF() do{ OUT_WRITE(PS_ON_PIN, !PSU_ACTIVE_HIGH); powersupply_on = false; }while(0)
> TO
> #define PSU_PIN_ON() do{ OUT_WRITE(PS_OFF_PIN, PSU_ACTIVE_HIGH); OUT_WRITE(PS_ON_PIN, PSU_ACTIVE_HIGH); safe_delay(1000); OUT_WRITE(PS_ON_PIN, !PSU_ACTIVE_HIGH); powersupply_on = true; }while(0)
> #define PSU_PIN_OFF() do{ OUT_WRITE(PS_OFF_PIN, !PSU_ACTIVE_HIGH); powersupply_on = false; }while(0)
> 
> This should allow you to do a M80 to turn on the relay and then a M81 to turn off the relay.
> If you noticed in the code the M80 does not keep the PS_ON_PIN high the entire time, it goes high and then waits a second and then goes low. This is to simulate a 'button' press.

Thanks for this info! If I knew it'd be such a hassle, I would've bought the old version 1.0.

Sucks that the source wasn't released for the firmware for v1.2

Zuckme, 2020-11-29:
What a shame!!! Why deliberately throw in the trash can the M80? And it looks like the old version 1.0 was supporting it?
ehhhh? Who was the good smelling manager taking this decision while the RnD department was banging their heads against the  lab benches? And you used a 
Not a happy customer here.

Anyway I did a little bit of reverse eng.
This is the short circuit detection:
[image]
If the 5V_IN pin is below 10% of 5V (or better said the Vcc voltage after the ACmain DC5V converter) it triggers. So it is 4,5V.
BTW U3 is a U3 is a LMV358 Op Amp.

I could get the M80 working with some hw modification but the right way is a

> 
> 
> Need a STC15 firmware modification ?

as already mentioned.

Now I go back at my bench and try to find a nice solution to this circus....

Zuckme, 2020-11-29:
Done! Took my about 2 hours to get all the details..... it was an easy reverse eng.

I never saw something like this... this board is .... ehhm ehmm... funny.

**1) Activate the M80**
well well this was easy... but you need the right tool. You need to be able to rework SMD components on the board.

what you need to do is:

- Desolder and remove that poor STC15 µC which is doing not a clever job programmed like that.
- Bridge the Pin P3.2 with P5.5, see below

[image]

and magically M80 and M81 are now supported..
You loose the LED on or off and the short circuit detection.

Here a shot during the rework:

[image]

I suggest to use a coin to shield the AD/DC converter, kepton tape is nice so no 0603 componets wil fly away with hot air.

Here the bridge:

[image]

All golden and shiny? well not really.... please continue reading.

**2) Increase the impedance of the input pin**

Well this is technical but IMHO important point. 
Basically this is the S pin goes through R4 and to the base of [this](https://16748.lightstrade.com/view/516066/J3Y-NPN-SMD-Transistor-S8050.html) NPN transistor Q2. The collector of Q2 is connected to P5.5 of (U2) STC15 µC.
Problem is R4 is freaking 100 Ohm!!!!! 

[image]

Zuckme...... so basically this input is was draining 25mA from the Marlin µC output PS_ON_PIN pin, which is no bueno. You should stay below 20mA, way below.... I almost could not believe it. So 25mA current base to turn on a NPN? No thank you.... this is hilarious. This explain why I was measuring 2.7V at the PS_ON_PIN when the PSU was on.... with 25mA the 3.3V state high pin out voltage were going bananas....

So I swapped the R4 with 1KR (1KOhm) resistor, and voila... I was down to ca.... 3mA....

If you do not have that resistor laying around like me you could take the R13...

[image]

which is 1.8KR there only for the short circuit detection.... which is not functional anyway without the  STC15.
It should work but I did not tested it.

Here the finish (Zuk)hack:

[image]

If you don't know what I am talking about, just leave it.
Of course you can damage your board, if this happens just buy a v1.0 version.

Hey BIGTREETECH.... I am so sorry... but this is what you delivered to me...

Time to work on the V1.3...

piotrkochan, 2020-12-18:
@Zuckme do You think this modification will work with skr mini v2.0? currently relay doesn't respond to the m80/m81 so it's totally useless device (#15)

Zuckme, 2020-12-18:
I have a SKR Mini E3 v2.0. No problem.

olivbd, 2020-12-18:
Works fine on BTT GTR V1.0! Thanks @Zuckme 

[image]

piotrkochan, 2020-12-19:
> I have a SKR Mini E3 v2.0. No problem.

I only switched pins to the SG connector and it work without modification, still I'm going to replace 100 ohm resistor later

Zuckme, 2020-12-26:
> I only switched pins to the SG connector and it work without any modifications, still I'm going to replace 100 ohm resistor later

Mine had also the SG connector pins in the wrong position, M81 works without any mod/hack. For M80 you need to do some hacks.

rubino25, 2021-03-07:
> Done! Took my about 2 hours to get all the details..... it was an easy reverse eng.
> 
> I never saw something like this... this board is .... ehhm ehmm... funny.
> 
> **1) Activate the M80**
> well well this was easy... but you need the right tools. You need to be able to rework SMD components on the board.
> 
> what you need to do is:
> 
> * Desolder and remove that poor STC15 µC which is doing not a clever job programmed like that.
> * Bridge the Pin P3.2 with P5.5, see below
> 
> [image]
> 
> and magically M80 and M81 are now supported...
> Of course you need now an independent voltage source to turn it on and off. The marlin board should get power regardless of the PSU state.
> You loose the LED on or off and the short circuit detection.
> 
> Here a shot during the rework:
> 
> [image]
> 
> I suggest to use a coin to shield the AD/DC converter, kepton tape is nice so no 0603 componets wil fly away with hot air.
> 
> Here the bridge:
> 
> [image]
> 
> All golden and shiny? well not really.... please continue reading.
> 
> **2) Increase the impedance of the input pin**
> 
> Well this is technical but IMHO an important point.
> Basically this is the S pin goes through R4 and to the base of [this](https://16748.lightstrade.com/view/516066/J3Y-NPN-SMD-Transistor-S8050.html) NPN transistor Q2. The collector of Q2 is connected to P5.5 of (U2) STC15 µC, and the emitter is at ground. A common emitter configuration [here](https://en.wikipedia.org/wiki/Common_emitter), nothing special.
> 
> Problem is R4, wait for it... here it comes... it is freaking 100 Ohm!!!!!
> 
> [image]
> 
> Zuckme...... so basically this input is draining about 25mA from the Marlin µC output PS_ON_PIN pin, which is no bueno. You should stay below 20mA, way below.... I almost could not believe it. So 25mA current base to turn on a NPN? No thank you.... this is hilarious. This explain why I was measuring 2.7V at the PS_ON_PIN when the PSU was on.... with 25mA the 3.3V state high pin out voltage were going bananas....
> 
> So I swapped the R4 with 1KR (1KOhm) resistor, and voila... I was down to ca.... 3mA....
> 
> If you do not have that resistor laying around like me you could take the R13...
> 
> [image]
> 
> which is 1.8KR and is there only for the short circuit detection (see my post above).... which is not functional anyway without the STC15.
> It should work but I did not tested it.
> 
> Here the finish (Zuk)hack:
> 
> [image]
> 
> There is no need change any parameter on the Marlin SW even `#define PSU_ACTIVE_STATE HIGH` is correct.
> 
> If you don't know what I am talking about, just leave it.
> Of course you can damage your board, if this happens just buy a v1.0 version.
> 
> Hey BIGTREETECH.... I am so sorry... but this is what you delivered to me...
> Time to work on the V1.3...

Good job. Works perfectly!! thanks

Mordi1984, 2022-10-10:
I Managed it in an unusual way with my octopus Pro 426 and the Relay 1.2.
First i Installed it, saw that it won´t work, also with 5V in on the Relay Board.
I installed the Jumber and powered it up with my Pi over USB-C, without success that M80 works.

Then i degraded it and installed an SolidStateRelay with the Power-Pins from my Octopus Pro board.
Now It is much more quiet, more stable, and M80 and M81 works without any problem.

Thank you Bigtreetech for Junk in my Basement.
With that Configuration on the Chip, it is useless.

But why complicated when you can install a same priced Solid-State Relay.