bigtreetech/BIGTREETECH-Relay-V1.2 #8: Error in Relay V1.2

tomac01, 2020-02-21:
Hello dear BTT team,

I have recently purchased the Relai V1.2. Unfortunately I have to state that it does not work correctly.
According to the manual the "automatic shutdown after printing signal line" should be used to keep the relay on or switch it off in case of signal loss.

Unfortunately the relay always switches off after 10-15 seconds. It does not matter if there is 3V on the signal line or not. I was not able to prevent the relay from switching off.

I must therefore assume that there is a fault in the relay.

Could you please check and confirm this!

Evg33, 2020-02-21:
https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/2

jffabreu, 2020-02-21:
Read this:
https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/2

and you need 5V... not 3V...

tomac01, 2020-02-21:
Hi, thanks for the answers.

@Evg33:
I have read the post and tried everything described there. Without success!

Therefore I have excluded the source of error of the motherboard and have put the power directly on the signal line.

@jffabreu:
If the signals on the motherboard are "true", then there is always about 3v. But I have also tried the 5v.

No matter if 3V or 5V are present as signal, the relay switches off the power after about 10-15 seconds.

So:
- the error source Moatherboard is excluded and
- the power (3V or 5V) is given directly as a signal.

Result:
No difference. The relay switches off.

Therefore I assume an error in the Relay V1.2 itself.

tomac01, 2020-02-22:
Hello, everybody,

I checked my whole setup again and found that I made a big mistake:

I swapped +3v (P1.0 for me) and GRD on the board connector.

Of course, then no signal reached the relay.

After I made the MarlinCore.cpp according to the information from [#2](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/2#issuecomment-586579656) (thanks to bryan-pl) all works perfectly!

Sorry, for my wrong contribution and possible confusion!

closed 2020-02-22 by tomac01