bigtreetech/BIGTREETECH-Relay-V1.2 #9: SKR Mini E3 V1.2 - no PS_ON_PIN for pins file pins_BTT_SKR_MINI_E3_V1_2.h

psjshep, 2020-07-13:
I'm trying to work out how to install the BTT Relay V1.2 to be used on the SKR Mini E3 V1.2 board.

There doesn't seem to be any definition for setting which pin to use in the pins_BTT_SKR_MINI_E3_V1_2.h file .... or I'm missing something.... I see it on the V2.0 file (pins_BTT_SKR_MINI_E3_V2_0.h), as that board obviously has the PS_ON 2 pins.

I do understand that the V1.2 board doesn't have a PS_ON pair of pins, but I was hoping to use maybe us the SERVOS or PT-DET +5v & gnd?  But I don't see how to configure that.

Regards,
Phil

flatcapper, 2020-07-19:
I’m getting ready to install mine also. I’m  fairly sure it simply gets wired in between the psi and motherboard. Then you add connector from relay board and the p-on socket on the mainboard.

psjshep, 2020-07-19:
Can you elaborate on the connections you plan you use?

Phil

flatcapper, 2020-07-19:
For what the psu relay?

psjshep, 2020-07-19:
In the Marlin src boards files for the SKR Mini E3 V1.2 board, there is no definition anywhere for a PS_ON_PIN... but there is one in the SKT Mini E3 V2.0 file..... PS_ON pin is marked on the board (which the V1.2 version doesn't have) - PS_ON is GND & PC13

So, is it a case that I can use another GND & signal pin like E0-stop (GND & PC15) & create a #define in the 1.2 pins file ? - I won't be using these 2 for a filament runout detection, as the one I have needs 3 pins (GND signal & +5v)

Phil

flatcapper, 2020-07-20:
Hello Phil, I think somewhere we got our signals crossed. I was talking about the btt relay v1.2 automatic off module to turn off my printer after done printing. I have both the btt skr mini e3 v2.0 and the tft35 e3 v3.0 installed on my ender 3 pro. They worked together perfect out of the box. Mostly a plug n play package. All I did was level the bed and started printing things. Never had any issues until I decided to try and install that counterfeit Uptanium abl system. It was designed for a direct connect to the stock ender 3 board and monitor. How I found out it was counterfeit was that I googled Uptanium and found their website and contacted them about using their abl system with the skr mini e3 v2.0 board. That’s when the owner from Uptanium told me that if I didn’t purchase the abl package from Amazon then it was counterfeit. He tried to help me even though the item clearly wasn’t from them. However after he started talking about a bunch of changes to the firmware I got lost. I’m still new to all of this. I very, and I mean very mechanically inclined, however after my accident back in 2015 when I fell and fractured my neck, I must have hit my head harder then I first thought. I’ve noticed little things over the last 5 years, however whenever anything complicated or requiring hard thought processes my brain just starts to lock/fog up, then things don’t make any sense. I did manage to screw up the auto home on my printer as the monitor has a page for bed leveling right there for setting up a bl touch. Somehow I saved a setting that placed my home in a different place. Also with the abl sensor connected to my z stop port I think it wasn’t reading correctly. I wound up pulling the abl package off the printer and re-installed the z stop switch. Then with some YouTube help I was able to find the updated version of Marlin for the skr mini e3 v2.o firmware, as well as the tft35 firmware. I got them both “flashed” to the motherboard and monitor and everything is working once again. I now ordered a bltouch clone from Aliexpress, however I think it will be much easier to install since the btt boards are laid out to work with the bltouch type of bed leveling sensors. I also know where the firmware file is for setting up the bl touch sensor also. So I’m hoping when it arrives I’ll be able to get it installed.
Sadly my ender 3 pro still isn’t centered when It prints anything. I’m not sure why its off center but it is and I don’t know how to fix this. I’m not good with all this g code stuff, but I should be able to figure it out if I can find a good guide to learn it. I asked Btt if they had a wiring diagram for the relay v1.2 auto shut off package. I saw one somewhere and now can’t find it. I will do some more looking. Also I’m not sure if I should connect the relay to the motherboard, or the printer board. They both have ports for the auto shutoff. If your still needing help with what to connect where I will send you the link on wiring this when I get it.
Just to set things straight, I’m using the btt skr mini e3 v2.o motherboard, and your using the skr mini e3 v1.2? I’m not sure what the big differences are between them other then the v2.0 has the extra z axis port and I think it has the bl touch ports also. Did I mention that I had the btt smart filament sensor also? I connected it to my printer port that is designed for it and then just turned it on in the printer options for filament sensor.
Well feel free to get back to me if you still in need of any help. You may in fact end up helping me more then I’m able to help you. I’ll start installing the auto shut off tomorrow as I’m to tuckered out tonight I just got the printer printing again and the my little pony is in the works that was supposed to be for my grand daughter yesterday for her birthday. Next is to try and get abs to print on my ender. I may just put the all metal hot end in first as it will easily handle the higher heat. Sadly I may have to upgrade the bed for running higher temp filaments. Otherwise I’ll just use the ender 3 for printing pla.
Again I hope I was able to help in any way and if you need anything just let me know.
Thanks,
Perry

psjshep, 2020-07-20:
Hi Perry...
       nope.. wires not crossed.... I am talking about the BTT relay V1.2... so I can shut off the printer after finishing a print.

The logic connection from the printer's motherboard needs to have a 2-pin "PS_ON" - the Marlin firmware is then tweaked so that the power off logic is enabled (it's disabled by default) and the required pin is defined - this is the PS_ON_PIN variable defined in some of the config files...

The SKR Mini E3 V1.2 (yeah.. confusing that both the relay & motherboard are V1.2) does not have a specific PS_ON pin, but there are other pins which possibly could be used if not being used for other stuff - the E0-Stop for example.... but the 1.2 motherboard config files in the Marlin source doesn't have a variable defined for PS_ON_PIN.... hence my dilema.

I hope that makes it a bit clearer?

Sounds like you had a NIGHTMARE with your original stuff... keep us posted here on how you get on...

Phil

flatcapper, 2020-07-20:
Hello Phil, I think that's ine of the nice things about the akr v2.0 board is that it has a port for said ps_on as well as.the tft35 board. I think I'm going to just plug the relay into the tft35 board.

psjshep, 2020-07-20:
Good luck Perry... let me know if you think I may be of some help if you need to modify the firmware...

Phil

flatcapper, 2020-07-20:
Hey Phil,
Did I mention that btt wrote me back and sent me the github link to the instructions for the relay v1.2? The instructions also included wiring guide what to connect where. I'm also going to install a momentary reset switch to reset the relay. I noticed in the instructions that there was two pins present for this feature. Also I was able to modify one of the on/off switches into a momentary switch. it's nice as I had plenty of extras since they were like 12 for $6. Since I printed out the rear enclosure model from teaching tech, I have plenty of room to mount the relay in the raspberry pi section side, complete with mounting screw points.I may take you up on the offer of help with the firmware. I'm planning on buying one or two more printers if the stimulus money comes through. One basic like ender 3 v2.0 and something on the higher end that comes with auto bed leveling, and can do stock ,pla, abs, petg, tpu, carbon, nylon. Not sure what can do all those but will start looking. I still may purchase a Tarantula pro rs if they have a decent.price and are available. Tried to order one before I got my ender 3, but that company has some issues, like terrible customer service, and an issue with not saying whenever a product is in stock and when.its not. But that's an entirely different story. Well I hope you figure out how to get your relay hooked up. Perhaps you could get a tft35 to go with your skr v1.2 board and just plug the relay into the port in the monitor.
Perry

psjshep, 2020-07-20:
Yeah.... I considered ordering a TFT - maybe the cheapest option rather than going for a v2.0 board....  I'll put it on hold for the moment as I'm about to work on a filament-out mod next.... that's more important at this time...

Thx for info...

Phil

BenoitGrou69, 2020-08-14:
Hi! I've been bothering to configure the V1.2 module on my SKR mini E3 V1.2 for 24 hours and I just saw this tiket. My module works correctly, I just need to manage the 5V. I connected the S&G on the PC12 of the PT-DEC but without result.
How did you do ?
Sorry for my shitty english

psjshep, 2020-08-14:
I am still trying to work that out but haven't had any feedback from BTT...

(your English is just fine.... better than my French is :) )

flatcapper, 2020-08-14:
Well not sure if you got my last update on the relay module. After BTT sent me both pictures of what to connect where. All good as after connecting things to the correct port on my tft35 when I power up the printer it stays running. So I decided to try and run a print. when my hotbed started to heat up, my printer reset itself.  I'm thinking its loosing power and the module is resetting.  It's not shutting down and staying off like the module is supposed to do, it's just rebooting the printer. Have no idea why. So I ripped the module out. I'm done trying to make these accessories work. There should be no reason why the end user has to jump through all these hoops to.make a accessory work. So good luck to you all I'm planning on just selling mine on eBay.
Perry

psjshep, 2020-08-14:
I have to agree with you, Perry....

I am a seasoned IT analyst of over 35 years..... so GAWD only knows how they expect a normal enthusiast has any chance of succeeding with this!

Good luck...

Phil

flatcapper, 2020-08-14:
Phil, I'm having the same issues with my bltouch. Cant seem to get it to work right either. Part of the fault is my choice to install a btt skr mini e3 v2.0 mainboard and tft35 to improve the performance, however this now creates a landmine that I have to walk through in trying to recompile the software. I'm going to play with it some more today and if I cant get it to work then it's getting ripped out also. As I sId there is no reason that it should be this difficult for end user to install such an important upgrade.
I used the pre configured firmware from btt for the bltouch on my ender pro. Sadly because not every printer is set up the same, which makes this preconfigured firmware not work correctly. Which leaves me with a broken printer. I've had a fellow 3d printer enthusiast trying to help me with the firmware but doing this remotely by messenger is tough. so at least before with manual bed leveling I was able to print things. Right now my printer has been sitting idle for 3 weeks. I've been tempted to smash my printer from frustration.  I was loving my printer after installing the btt mainboard and tft35 with whatever firmware came on it in the box. it worked perfect. well I need to go try to figure out all of this. Hope your printer is working how you want it to.
Best wishes,  Perry

psjshep, 2020-08-14:
I do understand... I have my bltouch working.... but still to get the runout sensor to a stage of confidence that I can make a code change...

I may just leave it with what I have & err on the side of caution when the spool gets low.... therefore bypassing the need for a runout sensor.....

The Ender 3 pro is a low end 3d printer... I suppose we should remind ourselves of that.... QED

Phil

BenoitGrou69, 2020-08-17:
> Phil, I'm having the same issues with my bltouch. Cant seem to get it to work right either. Part of the fault is my choice to install a btt skr mini e3 v2.0 mainboard and tft35 to improve the [...]

I have the SKR mini E3 V1.2 with the TFY35 screen and I had no problem completing the firmware with the BL Touch

Timbonade, 2022-09-25:
Hey, were you able to solve the problem?

flatcapper, 2022-09-26:
That is a very long-winded answer. Yes I got it configured and working. I printed a lot of stuff up until some little mossfet burnt out going to one of the axis motors. The board was less then one year old, only used in my Ender pro and even then not near as often as one would imagine. I had read that btt boards had a 1 year warranty against failures. After contacting them I was told to go fly a kite. That warranty was only on their factory systems, not individual components. They said something about not offering a warranty as they had no control of how individual components were installed. But I had won this other newer free board from btt and tried to install that board with no joy. So now all 3 of my 3d printers have been collecting dust for months. Perhaps this winter I'll get the printers going. I purchased a new skr minie3 v2.0 board, I just haven't installed it. I did notice that btt increased the size of the mossfets on the v2.0 boards. I'm also now looking at the world of benchtop lasers for doing some leather etching..

Timbonade, 2022-09-26:
So I bought the BTT SKR mini E3 V1.2 board because it made my Ender 3 much quieter and I can use things like BL Touch and the TFT 35 touch screen.  I don't use the printer every day either, but so far I'm very satisfied with the boards.  I solved the problem with the Psu Auto off relatively quickly.  If you need help feel free to send me a message.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.