bigtreetech/BIGTREETECH-Relay-V1.2 #11: PIN 5V and RST on v1.2

nictronik99, 2020-09-11:
What is the use of those two 5v and RST pins on version 1.2? In the instructions it says that they should be used to reset or reboot but it doesn't work. Maybe I'm wrong, someone can tell me how How to reboot using those pins?

combolek, 2020-09-17:
This only works once the relay is in the power off state. Shorting the 5V and RST pins together resets the relay and turns the power back on.
It will not do reset/reboot if the power is already on.

combolek, 2020-10-07:
I am not sure. There are a couple of GPIO plugins for OctoPrint but I don't know if they can do exactly what you want.
I use Klipper which has software restart. In cases when that is not enough I just do the power off/power on sequence via GPIO. So I don't use or need the reset button on the board.
I do have an external power button (in addition to RPi) wired up to the RST and 5V on the relay just so I can turn on the printer with a physical button. But it's not a reset, it doesn't do anything if the printer is already on.

Zuckme, 2020-12-04:
> do you know by chance if there is a plugin on octopi to connect the gpio to the rst button and turn on the printer

You need to get the board to support the M80 command.
More details [here](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/7#issuecomment-735447011)