bigtreetech/BIGTREETECH-Relay-V1.2 #12: BTT Relay V1.2 on BTT SKR Mini E3 V1.2 Installation

Timbonade, 2020-09-27:
Hello, I have now read through some error descriptions, but I am not so sure how to connect the relay to my SKR Mini E3 V1.2.
I installed a touchscreen and a BL Touch.
Where should I connect the relay now?
And which pin is it then, I have to know that in order to indicate it in the firmware.

Sorry for my bad English, I'm from Germany.
---------------------------------------------------------------------------------------------------------------------------------------

Hallo, ich habe mir hier jetzt einige Fehler beschreibungen durchgelesen, doch so ganz sicher bin ich mir nicht, wie ich das Relay jetzt an mein SKR Mini E3 V1.2 anschließen muss. 
Ich habe Ein Touchscreen und einen BL Touch verbaut. 
Wo sollte ich jetzt das Relay anschließen? 
Und welcher pin ist es dann, das muss ich ja wissen um es in der Firmware an zu geben.

th0547, 2021-01-26:
Hello!

maroca2177, 2021-03-23:
> ¡Hola!
> 
> También lo estoy instalando en un SKT mini E3 V1.2 y lo tengo instalado, solo necesito arreglarlo apagándolo después de ~ 30 segundos. Compartiré con ustedes cómo lo he instalado, pero lo haré un poco gráfico para que sea más fácil de entender. Lo publicaré mañana

Puedes mandar como lo instaslate?? En esa placa por favor

maroca2177, 2021-03-23:
Puedes mandar como lo instalaste por favor, gracias

vyacheslav-moiseev, 2021-04-07:
> Hello!
> 
> I'm also installing it on a SKT mini E3 V1.2 and I have it installed, I only need to fix it turning of after ~30 seconds. I'll share with you how I have installed but I'll make it a bit graphical so it's easier to understand. I'll post it tomorow
Hi ! 
Were you able to hook this up?