bigtreetech/BIGTREETECH-Relay-V1.2 #15: Does not work with SKR mini e3 v2.0

piotrkochan, 2020-12-15:
I'm struggling with connection to the SKR Mini E3 v.20, I'm just impressed how it could not work with same manufacturer devices but ok,
I connected +5V and GND from the SKR SPI Connector to the BTT Relay +5 and GND.
I also connected PS-ON to the "SG" connector next to 5V_IN and GND (center one).

Before that I flashed Marlin firmware with following options:

```
#define PSU_CONTROL
#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_STATE HIGH 
```

After wiring everything powered on but after +- 10 seconds relay clicks and printer powers down. 
I've tried to set PSU_ACTIVE_STATE to LOW but same effect, so I reversed it back to HIGH.

After some troubleshooting I've moved PS-ON connector to next  "SG" on the Relay board and now printer is always ON, but commands M80 and M81 have no effect at all so whole thing is useless.

Is there anyone that had similar issues and know how to resolve them?

Zuckme, 2020-12-18:
> I connected +5V and GND from the SKR SPI Connector to the BTT Relay +5 and GND.

This is wrong, you don't need to do that.

piotrkochan, 2020-12-18:
> > I connected +5V and GND from the SKR SPI Connector to the BTT Relay +5 and GND.
> 
> This is wrong, you don't need to do that.

I had to reverse connector in order to make it work.

closed 2020-12-18 by piotrkochan