bigtreetech/BIGTREETECH-Relay-V1.2 #16: Momentary power button on/off

BDiff, 2020-12-25:
Would it be possible to change the firmware so the momentary switching of 5v and RST turns the relay off as well as on? I'm currently working on a printer where board had a built in relay for on/off. The momentary button turned the printer on and off. We have replaced the old printer board with an SKR Pro 1.2 and I'm using the BTT Relay 1.2 with the existing momentary button to turn the printer on. It would be more favorable to be able to shut it off as needed.

Zuckme, 2020-12-26:
I doubt they will ever change the firmware for your needs, I suggest to rev eng the board and hack it, check there #7

BDiff, 2020-12-27:
While I totally agree with you on that, I have two more printers to adapt an SKR Pro and relay board to. I won't be the end user of the printers, so I'd like to be able to use the relays as they are, but with different momentary button logic. I'll check out the hack and see what it requires. Thank you.