bigtreetech/BIGTREETECH-Relay-V1.2 #17: Relay resets itself

Cabinfourus, 2021-01-13:
I installed the BTT relay v1.2 onto my skr 1.4 turbo. I used the wiring diagram from issue number five except I have my 'power off' connected to the E0/Z max pin (1.0) and the 'power loss' to the E1/Y max pin (1.25). My printer will finish the print and shut off as expected, using the M81 G code at the end of my script. After about 15 seconds I hear the relay click and power is returned to my printer like normal. I want to keep it off after prints. I am using Simplify3D software and printing from a USB connected PC.

Also, I tried installing the 12IC EEPROM and failed. I followed the very lacking tutorial the provide and was never able to get it working. I kept getting an error for data size mismatch. I tried all the M500, 501, 502 combinations and nothing seemed to work. 

I am running the SKR 1.4 T with TMC 2209 (sensorless homing) I did not remove the diag pin. 

Any ideas?

Zuckme, 2021-01-26:
Did you configured and flashed the Marlin FW correctly?

NezSez, 2021-03-28:
@Cabinfourus I can't help you with your relay issue, but you should adjust settings according to the post from thisiskeithb in this issue [Marlin 21356](https://github.com/MarlinFirmware/Marlin/issues/21356) for the EEPROM.  I have it working on the SKR 1.4 Turbo, and upgraded the chip to a 24C512 and it is working.