bigtreetech/BIGTREETECH-Relay-V1.2 #19: How to upload FW on Relay v1.2?

davbcr, 2021-02-17:
Sorry to bother, but i didn't find any word about how to upload preconfig FW on the realy module.
Can you please adress me in the right direction?
I wonder if i can use my CH340G module to do it.

combolek, 2021-02-17:
Do you really need to? The relay comes with firmware pre-installed. Do you have an alternative firmware that you want to use?

If you really want to then yes, you should be able to use CH340 module to program it. You will need programmer software like this: http://www.stcmicro.com/rar/stc-isp6.86e.rar. Set the chip type to STC15W201S. But this is not for the faint of heart, there is a lot of other settings. I have NOT done it myself.

Also, keep in mind that there is no way to read the existing firmware from the chip so it's a one way street. BTT posted a firmware hex file here on github but there is no guarantee that this is the same as what is actually shipping.

davbcr, 2021-02-18:
Perfect i just wanna this conformation not to proceed on any damage. Thanks a lot.

closed 2021-02-18 by davbcr