bigtreetech/BIGTREETECH-Relay-V1.2 #20: Relay placement in line if modules.

agenttatdaddy, 2021-03-02:
Good day friends,
I am in the process of doing a complete overhaul of my Ender 5 Plus. I am going with the SKR E3 Turbo, 24v BTT UPS, and very large hotbed MOSFET (I didn't realize the size when I ordered it), Smart filament sensor, and of course the BTT Relay 1.2. I am thinking the power should go to the UPS first, then the relay, then the MOSFET and mainboard. That way the UPS does not kick in when the relay shuts the printer down. Any advice is greatly appreciated. This is my first complete overhaul so any good tips will help, such as good ways to completely enclose the relays for safety reasons. I also plan on using an LED circle light on the printhead, so I will also be using the 5v adapter for the SKR E3. I hope that helps. 
[image]

NezSez, 2021-03-28:
@agenttatdaddy I am also installing the relay v1.2 and mini UPS on an E5+. How did your plan with the order work out? By putting the UPS first in line from the power, did you avoid the frequently reported problem of having to wait for the caps to drain before the printer would actually shut off when you switch the PS switch to off manually? You didn't mention the display you were using, and some have had issues depending on whether they connected the modules to the mainboard or the BTT TFT screen, which can also affect if they work depending on which mode (Touch or simulated Marlin(12864lcd)) they start prints in, etc.

ssheytanov, 2023-01-05:
Any update here? I want to achieve the same thing as OP, I have a TFT35 and I'm wondering wether to connect them to that or to the MB? I print mostly from the TFT more and occanionally start prints from the ESP3D on my ESP01S connected to the MB. If I print new case for the TFT35 I can move the ESP to the TFT as well.