bigtreetech/BIGTREETECH-Relay-V1.2 #21: Official BTT Relay V1.2 schematics available

combolek, 2021-03-28:
This is not an bug report but I got the Relay 1.2 schematics from BTT and a permission to share it, so here it is. It probably has been already reverse engineered by many people by now but still.
[Relay V1.2.pdf](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/files/6216849/Relay.V1.2.pdf)

vale075, 2021-04-01:
I created a pull request in order to add it, the doc for this is quite poor...

flo199213, 2023-05-23:
[image]

Thank me later...
I don't know how many times someone has had to translate that for themselves

Escrich, 2023-08-21:
> [image]
> 
> Thank me later... I don't know how many times someone has had to translate that for themselves

Relay contacts in schematic must be wrong, following the schematics, the relay must be switched on to shut down power, and must be on to maintain the printer off, sorry, but the normal way to work is to have the relay energized, and shut it down to shut down the printer

Escrich, 2023-09-08:
Ok, assuming the relay must be energized in order to shut down the printer, from my point of view, there is a mistake on the schematics, related to the zone marked as Power failure detection, please, look at Issue #34