bigtreetech/BIGTREETECH-Relay-V1.2 #33: manta m8p and klipper

jariddler, 2023-01-13:
klipper loads to slow for the relay to get the pin high. so the btt relay v1.2 triggers a shutdown.  make shift solution is to hold the reset button until klipper loads then the pin is high and no shutdown is enacted. there should be a firmware available for the relay to have a long delay so klipper has enough time to set the pin high.

Magic2991, 2023-01-21:
I have the same problem.  there is a problem with some hardware configurations. I think it would be better to change the relay so that it only needs one signal to disconnect the power. means as long as no voltage comes in. will not be switched. so no signal should be present at the start and klipper3d etc can start in peace.

jariddler, 2023-01-21:
> I have the same problem. there is a problem with some hardware configurations. I think it would be better to change the relay so that it only needs one signal to disconnect the power. means as long [...]

I did another work around. I just rebuilt my firmware for my manta except this time I set the pin high in the configurations menu and that worked.

MR-CRAW4D, 2023-01-22:
Setting the PS_ON pin to high only works if your not using the SPI2 3.3v for an accelerometer. PC3 is tied to SPI2 and setting it high at boot will Ignore any spi2 communication because PC3 needs low to listen for instructions.

A longer term and reliable solution is connecting to a 5v fan pin that is not shared to SPI. 

KLIPPER
1. Set board jumper for 5v on fan3 (4th fan socket from 
    power block)

3. Set PE5 (fan3) to high in menuconfig using GPIO Pins
    (Do not add ^ or ! )

4. Add to printer.cfg:

########################################
# POWER ON/OFF
########################################
[output_pin psupower]
pin: PE5
value: 1
shutdown_value: 0

[gcode_macro M80]
gcode: SET_PIN PIN=psupower VALUE=1

# power off
[gcode_macro M81]
gcode: 
    SET_PIN PIN=psupower VALUE=0

[gcode_macro SHUTDOWN]
gcode:
    M81
    {action_call_remote_method("shutdown_machine")}

[gcode_macro REBOOT]
gcode:
    {action_call_remote_method("reboot_machine")}

 (Must do)
Start Manta M8P from power switch. It will auto shutdown after 10-15 seconds. Restart from jumper on Auto Shutdown board (or momentary switch wired to jumper).

That's it!

jariddler, 2023-01-22:
> Setting the PS_ON pin to high only works if your not using the SPI2 3.3v for an accelerometer. PC3 is tied to SPI2 and setting it high at boot will Ignore any spi2 communication because PC3 needs low to listen for instructions.
> 
> A longer term and reliable solution is connecting to a 5v fan pin that is not shared to SPI.
> 
> KLIPPER
> 
> 1. Set board jumper for 5v on fan3 (4th fan socket from
>    power block)
> 2. Set PE5 (fan3) to high in menuconfig using GPIO Pins
>    (Do not add ^ or ! )
> 3. Add to printer.cfg:
> 
> ########################################
> 
> # POWER ON/OFF
> ######################################## [output_pin psupower] pin: PE5 value: 1 shutdown_value: 0
> 
> [gcode_macro M80] gcode: SET_PIN PIN=psupower VALUE=1
> 
> # power off
> [gcode_macro M81] gcode: SET_PIN PIN=psupower VALUE=0
> 
> [gcode_macro SHUTDOWN] gcode: M81 {action_call_remote_method("shutdown_machine")}
> 
> [gcode_macro REBOOT] gcode: {action_call_remote_method("reboot_machine")}
> 
> (Must do) Start Manta M8P from power switch. It will auto shutdown after 10-15 seconds. Restart from jumper on Auto Shutdown board (or momentary switch wired to jumper).
> 
> That's it!

My way works just fine for me no issues at all.

Magic2991, 2023-02-07:
hello, can you please post a guide for ponit 2. how to do set the pin PE5 too high via makeconfig and gpio. Thank you

cjhmdm, 2023-02-26:
> I did another work around. I just rebuilt my firmware for my manta except this time I set the pin high in the configurations menu and that worked.

which pin did you set high, PC3?

Marck80, 2023-03-31:
> Setting the PS_ON pin to high only works if your not using the SPI2 3.3v for an accelerometer. PC3 is tied to SPI2 and setting it high at boot will Ignore any spi2 communication because PC3 needs [...]
What version of M8P do you have? with v1.0 should it be linked here?
[image]

jariddler, 2023-03-31:
Been working great for 2 months now

Marck80, 2023-03-31:
what configuration do you have? m8p? driver? adxl345?bltouch?
thanks

jariddler, 2023-03-31:
Manta m8p 
TMC2209
Btt sb2209 canbus
Utoc 3
Pitft70 touchscreen

Marck80, 2023-03-31:
with 2209 you don't use the spi but the uart so you have the free spi bus. I use the 5160 with spi unfortunately :(

talv2010, 2023-10-14:
I did same firmware with delays. You can use it.
https://github.com/talv2010/biqu-relay-v1.2-firmware-modified

Griborim, 2023-10-31:
> I did same firmware with delays. You can use it. https://github.com/talv2010/biqu-relay-v1.2-firmware-modified

The firmware is not bad.

Unfortunately, it no longer has a short circuit detection feature.
If you knew the inputs you could make the firmware yourself, but unfortunately that is not documented.

Alexfx82, 2024-02-02:
> Setting the PS_ON pin to high only works if your not using the SPI2 3.3v for an accelerometer. PC3 is tied to SPI2 and setting it high at boot will Ignore any spi2 communication because PC3 needs [...]

Hello! sorry for the stupid question, I'm just learning kipper! Can you tell me where to insert all these lines into which file? Thanks