bigtreetech/BIGTREETECH-Relay-V1.2 #34: Documented schemes for everyone; and a tough about an error on them, (maybe)

Escrich, 2023-09-08:
Dear all, from my point of view, there must be a mistake in the scheme, because looks like if input marked as 3, were badly defined or drawn, please check it and come back me with an answer, I think the rest of schematics and explanations could help some other people

[image]
[image]

Escrich, 2023-09-09:
También una copia con las instrucciones para castellano hablantes, que fue la primera que se hizo:

[image]

Escrich, 2023-09-15:
Anyway the complete unofficial document created:

[20230908 Alimentación rele BTT V1.2 (unofficial document).pdf](https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/files/12619026/20230908.Alimentacion.rele.BTT.V1.2.unofficial.document.pdf)

tiagomaricate, 2024-02-02:
Hi @Escrich !
I posted a text asking for help in another topic (https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/issues/4#issuecomment-1924661118) 
Maybe could you check that?
Looks like you understand very well about the relays, so maybe you can give me a little help.

Escrich, 2024-02-03:
Need to talk about